HelTecAutomation/Heltec-E-Ink #5: Hi, the code for Heltec 2.13 inch does not compile!

chetan-smart, 2019-04-29:
There are function prototypes missing. Can someone look into this?

solotaker, 2019-05-01:
> There are function prototypes missing. Can someone look into this?
Hello, 
I tried to compile this library and didn't find the problem you said.
Do you give me more information about the compilation error?

chetan-smart, 2019-05-02:
I uncommented ESP8266 here:

https://github.com/HelTecAutomation/e-ink/blob/master/src/imagedata.h#L2

And I uncommented 213 here:

https://github.com/HelTecAutomation/e-ink/blob/master/src/imagedata.h#L8

I then compiled the code.

The problem is that USE_213 is not implemented and does nothing at all. 

Refer to here:

https://github.com/HelTecAutomation/e-ink/blob/master/examples/e_ink/e_ink.ino#L97

And when I try to copy the implementation of USE_154 into USE_213 in e_ink.ino, I get a lots of errors because USE_213 has a lot of missing function prototypes.

So in conclusion, the main problem is this:

The implementation of USE_213 in e_ink.ino is incomplete and does not do anything sensible.

solotaker, 2019-05-05:
> I uncommented ESP8266 here:
>
> https://github.com/HelTecAutomation/e-ink/blob/master/src/imagedata.h#L2
>
> And I uncommented 213 here: [...]

I tried to use WIFI_KIT_8 (ESP8266) to drive the 2.13 inch E-ink. The e-ink works fine without any exceptions.

I think your problem may be caused by the following reasons:

1. You may not have properly installed the ESP8266 build environment package;
     Please refer to here: [](https://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino) 
2. Check your SPI and other function pins.
On WIFI_KIT_8:
                               D/C                        GPIO4 (D4)
                               SDI (MOSI)             GPIO13 (D7)
                               CS                           GPIO15 (D8)
                               CLK                         GPIO14 (SCL)
                               BUSY                      GPIO5 (D1)

closed 2019-05-09 by chetan-smart