HelTecAutomation/Heltec-E-Ink #8: No really usable examples

martinberlin, 2020-03-24:
Sorry but I have to mention that the implementation examples compared to what we can see in another libraries like [gxEPD](https://github.com/ZinggJM/GxEPD/tree/master/examples)

I bought one of this since I want to give it a try to another type of displays but I'm greatly surprised to see this. Is not possible to run this display using gxEPD?
In my first try I still couldn't but I did not try so hard, just wanted to have some feedback.
Thanks in advance!

v-blazhko, 2020-03-27:
I'm also struggling with runnig this display with my ESP32.

martinberlin, 2020-04-05:
Actually I've read that people managed to drive this display also with gxEPD but I tried without any luck. So I will go back and try with this examples instead probably will come back here to report. 
Heltec, please! If you do a library for one Eink at least add examples of how to do stuff with it. 

I really don't understand what is the point of doing it like this when other libraries offer a lot of usable examples (Like the one I mentioned)

xdubx, 2020-04-06:
I make a video to it :D 
https://www.youtube.com/watch?v=CFhFiNvTUjI

martinberlin, 2020-04-07:
Very cool @xdubx,  I will try this afternoon!

v-blazhko, 2020-04-09:
Thanks @xdubx! Do you have any ideas what pinout should be used with ESP32 with the library you mentioned!

v-blazhko, 2020-04-09:
@martinberlin I'm also trying to run it on ESP32, did you get any success with gxpEDP library?

martinberlin, 2020-04-18:
Not yet. I don't think I will loose any more time with this though...
My try was using same class as @xdubx 

  #include <GxGDEP015OC1/GxGDEP015OC1.h>

And pins, but could be any, only ones that are used by default is SPI in esp32 MOSI-> 23 CLK ->18

int8_t EINK_CS = 32;
int8_t EINK_DC = 27;
int8_t EINK_RST = -1;
int8_t EINK_BUSY = -1;

It did not move a thing. Is still printed the Heltec automation thingie

martinberlin, 2020-04-19:
@v-blazhko I took some time this morning to put some order in this and forked the library. For me now it works:
https://github.com/martinberlin/e-ink

Please try to install my fork and let me know if it works. Just delete the old lib, change the include in platformio, and use the build_flag methods.  If you use arduino is also no problem, but just add defines before inserting the library itself in the main file. 
Let me know if it worked for you.

closed 2020-04-23 by martinberlin

1kc, 2021-06-25:
> @v-blazhko I took some time this morning to put some order in this and forked the library. For me now it works:
> https://github.com/martinberlin/e-ink-platformio
> 
> Please try to install my fork and let me know if it works. Just delete the old lib, change the include in platformio, and use the build_flag methods. If you use arduino is also no problem, but just add defines before inserting the library itself in the main file.
> Let me know if it worked for you.
> 
> UPDATE: Now that I've went a bit deeper in the gxEPD SPI communication I think a have a clue of what's going on. The LUT initialization look up tables are not being set exactly like Heltec and that maybe the reason that it does not refresh in ESP32. I don't know why it works on ESP8266 but I tried using same setup using same driver as @xdubx in his video and does not work for me. Also the controller in this Eink is IL0373 as per Heltec tech spechs so my best guess would be to copy one of gxEPD classes and fork it with this LUT Heltec initialization. Then there is a chance it works for this Eink and we can just use gxEPD with it's full support for other stuff and handy functions to handle bitmaps and other beauties (Like support for U8G2_FOR_ADAFRUIT_GFX Fonts)
> 
> By the way Heltec if you are reading this and you like my modifications I can make a pull request and delete my fork, just ask.

Thanks for your fork. This is the only way that works for my ESP32 + 2.9 Inch display.