HelTecAutomation/Heltec-E-Ink #9: Messy Lib

xdubx, 2020-04-03:
The display BW 152 is completely missing in the lib. ORDINARY_SCREEN not work for me << i have a 1.54 inch screen.

Pin declarations are strange 
`SPI.pins(CLK_PIN,MISO,SDI_PIN,CS_PIN);`
OK we have custom pin declarations but we use the standard MISO pin. Makes no sence.
Select.cpp have fancy defineds to.
the function Display_String dont work at all if you have not the right display.
lib.properties file have only avr architectures.

Can you take a look at the lib and rewrite it?

martinberlin, 2020-04-18:
This library is completely unusable! I already mentioned it in the other issue. 
Especially the intructions on how to set it up. I must say it's kind of a joke, lucky you could make it work with gxEPD and ESP8266. I had no luck with ESP32

martinberlin, 2020-04-19:
Dear @xdubx 
I took some time this morning to put some order in this and forked the library. For me now it works:
https://github.com/martinberlin/e-ink

Note that my display model is: USE_154_BW_GREEN 
I guess this name is because the Eink is covered with a transparent plastic with a green ending to remove.
Try it out and let me know if it works. Basically, just inject the library, using my fork and just add the build_flags like I wrote in the Readme example. Much easier, better readable, and keeping the variables outside the library environment.

iotleaf, 2020-05-08:
I am evaluating the heltec 1.54 board with green label attached, to run on an UNO board.
So far I have been unable to modify your library to work (spent 1 day on this).
Could you please configure your system, and test it, to run on an UNO, and then email me a zip file containing all the required files?.. My email is medical@iotleaf.co.uk
Thanks

martinberlin, 2020-05-09:
Sorry @iotleaf but I cannot do your homework. First of all I do not have an UNO at home at the moment. Have you tried the example I suggested using Platformio ?

iotleaf, 2020-05-09:
Hi Martinberlin, sorry - I posted this issue in the wrong place, the library I was refering to was Heltec, rather than your version for platformino

martinberlin, 2020-07-17:
If anyone is using ESP-IDF/ ESP32 I made a class that works with this display:

https://github.com/martinberlin/cale-idf/wiki/Model-heltec0151.h