HelTecAutomation/Heltec-E-Ink #11: To everyone having issues

zvoniimiir, 2020-06-05:
By now you know this repository is crap. This is what worked for me on a 1.54 e-ink display. It should work on other sizes as well.

NOTE: This display requires 3.3 data lines. It works on 5v data lines, but just barely. The display is not crisp. I've read it's possible to use a 10k/4k7 resistor divider for 5v data lines, but it didn't work for me.

Download this library https://github.com/ZinggJM/GxEPD

From Arduino IDE, open the example

Select this display
#include <GxGDEH0154D67/GxGDEH0154D67.h>  // 1.54" b/w

Use these pins
GxIO_Class io(SPI, /*CS=*/ 10, /*DC=*/ 9, /*RST=*/ 8); 
GxEPD_Class display(io, /*RST=*/ 8, /*BUSY=*/ 7); 

Connect as follows (display / arduino):
VCC / VCC
GND / GND
D/C / D9
SDI / D11
CS / D10
CLK / D13
BUSY / D7

martinberlin, 2020-07-17:
If anyone is using ESP-IDF/ ESP32 I made a class that works with this display:

https://github.com/martinberlin/cale-idf/wiki/Model-heltec0151.h

wimmatthijs, 2020-11-24:
Hi Martin,
don't know how your progress is on this issue, but i checked the Heltec library and there are 2 sets of SPI commands.
One set it is for the IL0389, the other, i have no idea....
Unfortunate because i'm trying to make it work.
Mine says E-ink Display V2 on the back.
According to the library this is a "special version" of the display....
Special being : even more badly supported than the original ha ha ha

Wim

wimmatthijs, 2020-11-24:
So i finally discovered that mine does have the SSD1681 however i couldn't get it to play nice with the library yet.

martinberlin, 2020-11-24:
Try to use my [ESP-IDF epaper library](https://github.com/martinberlin/cale-idf/). Maybe it's not such a quick start as the Arduino ESP32 framework, but my library definitivaly works :)
[image]

wimmatthijs, 2020-11-24:
I tried to give it a try, but it's not quite clear to me how to set up the 'hello worl' example for this.
Could you maybe reach out for 15 mins to give me a quick runthrough?
My appreciation would be very high.

martinberlin, 2020-11-24:
Sure one afternoon I can show you around, just send me an email to martin at luckycloud.de
But do you have ESP-IDF configured and can already compile something?

wimmatthijs, 2020-11-24:
Hi Martin, i wrote you an email, hope you got it.

In the meanwhile i figured out that my e-ink display actually has the SSD1681 installed,
and it worksperfectly fine with the GxGDEH0154D67 driver of the GxEPD library that can be found here : 
https://github.com/ZinggJM/GxEPD
The only thing i changed was the SPI frequency from 4MHz to 2MHz (as can be found in the original Heltec "library")

It is clear that there are indeed a couple of different versions produced by Heltec and it is not well documented at all which version you will get, other than the sticker indication.
So my guess is, that the people who are having the ones with BLUE sticker are having the IL0389 version?

Any way i'm happy to see it work now, unfortunately one is bricked/broken.
Hope i can get the tricolor 2.13" one to function as well...

Hope this info will help anybody strugling out there.

ps : Martin, i still want to try your library and have a small chat of course ;)

atis-s, 2021-03-30:
I have 2.13'' brw display, that I have tried countless libraries on, but it still displays that default image.. Is it possible I have damaged it by sending 5V data instead of 3.3v ?  Are there a way to tell ?

martinberlin, 2021-03-30:
I have a class in my ESP32 IDF component that works good with it:
https://github.com/martinberlin/cale-idf/wiki/Model-heltec0151.h

But I always had this version that you can see in my ESP32 watch that gas the IL0389 IC. Not sure if there are another versions out there. This is Espressif IDF, not arduino ESP32 but is easy to compile it if you have the IDF >= 4.2 installed

p4ent, 2025-08-02:
Hi everyone I'm trying to use an epaper GDEW042C37 BWY 4,2" sold by Heltec, and I'm using heltec e-ink board (https://heltec.org/project/e-ink-driveboard/)
no matter what library or model I use in the code, it seems I cannot send the text to the epaper. My guess is I'm using the wrong pins but I do not know what the real ones are

here is a code
```

// Alternative avec GxEPD2 (plus récente et maintenue)
#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>
#include <Fonts/FreeMonoBold9pt7b.h>

// Pour un écran 4.2" couleur (remplacez par votre modèle exact)
GxEPD2_3C<GxEPD2_420c, GxEPD2_420c::HEIGHT> display(GxEPD2_420c(/*CS=*/ 5, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4));

void setup()
{
  Serial.begin(115200);
  Serial.println("=== Initialisation avec GxEPD2 ===");
  
  display.init(115200);
  display.setRotation(1);
  display.setFont(&FreeMonoBold9pt7b);
  display.setTextColor(GxEPD_BLACK);
  
  display.setFullWindow();
  display.fillScreen(GxEPD_WHITE);
  display.setCursor(30, 100);
  display.println("HELLO E-INK");
  display.display();
  
  Serial.println("Affichage terminé avec GxEPD2!");
}

void loop() 
{
  delay(1000);
}
```

and what the serial monitor says:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x3a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1180
load:0x403c8700,len:0xc2c
load:0x403cb700,len:0x311c
entry 0x403c88b8
=== Initialisation avec GxEPD2 ===
Busy Timeout!
_PowerOn : 20001033
Busy Timeout!
_Update_Full : 20001029
Busy Timeout!
_PowerOff : 20001028
Affichage terminé avec GxEPD2!
```

does anyone knows how to fix please?